Spooks4576/Ghost_ESP #9: Not Operating

DefinetlyNotAI, 2024-05-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to the upload (use the firmware for version ESP32_V5)
2. Used PuTTY to connect (COM4, baud rate 14400)
3. See error:
`
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
`
`
wrong chip id 0x0002
`
`
wrong chip id 0x0002
`
`
wrong chip id 0x0002
`
`
wrong chip id 0x0002
`
`
wrong chip id 0x0002
`
`
wrong chip id 0x0002
`
`
wrong chip id 0x0002
`
`
ets Jul 29 2019 12:21:46
`

**Expected behavior**
the default console should be available

**Screenshots**
no need

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Opera GX
 - Version ...

**Additional context**
nope

Spooks4576, 2024-06-20:
Your Using the ESP32_V5 From Awok boards correct?

do you know exactly which ESP-32 your board has?

DefinetlyNotAI, 2024-06-22:
Oh, um... I am using the ESP32-WROOM board, as well as the dev board for the flipper, both do not work, i expect that for the wroom, but not the flippers esp32 board

Spooks4576, 2024-06-23:
Your Flipper Dev Board Should Work with 

https://github.com/Spooks4576/Ghost_ESP/releases/download/V1.4.6/flipper_dev_board.tar.gz

make sure you are flashing it correctly with the right offsets stated in the README

DefinetlyNotAI, 2024-06-24:
Ok now it works, what about the esp32 wroom?

Spooks4576, 2024-07-25:
depending on your board you could flash the minion marauder which uses the esp32 wroom

closed 2024-07-25 by Spooks4576

DefinetlyNotAI, 2024-07-25:
ok thanks