Spooks4576/Ghost_ESP #25: *Suggestion* 2.4 ili9341 tft support

DonAnonymousio, 2024-10-21:
little bit of a selfish want lol, but the 2.4tft is a pretty nice screen that shrinks down the form factor for devices a bit lol. just a little suggestion. koko's old hardware bin for the V4 works with the 2.4 tft nicely if that helps at all :).

Spooks4576, 2024-10-21:
2.4 is already supported i believe the resolution your talking about is 240x320 correct?

DonAnonymousio, 2024-10-21:
> 2.4 is already supported i believe the resolution your talking about is 240x320 correct? 

Yes that resolution. Just tried it out and the touch inputs dont match up with the screen inputs. It does however boot up and look very nice I may add.

Spooks4576, 2024-10-21:
> > 2.4 is already supported i believe the resolution your talking about is 240x320 correct?
> 
> Yes that resolution. Just tried it out and the touch inputs dont match up with the screen inputs. It does however boot up and look very nice I may add.

can you be more specific when you say touch inputs dont match up with the screen inputs does that mean the touch does not work at all? or is the touch just inaccurate.

Spooks4576, 2024-10-21:
closing this as the issue will be resolved in discord

closed 2024-10-21 by Spooks4576