Spooks4576/Ghost_ESP #29: compatability for the cheap yellow display? ( st7789 )

p1ssguzzler, 2024-10-28:


Spooks4576, 2024-10-28:
should be compatible if its a single USB port 

if not and it has a micro USB and USB C it is not supported and you will have to wait

Moroyi, 2024-10-28:
Feedback to compatibility: I am using an ESP32-24S032C CYD style board with a single USB-C ST7789 screen driver and GT911 touch driver.
Basic functionality is fine as in the software will run and perform actions as expected, if you can manage to select your wanted option; see below.
The screen could be flipped 180 degree. As is, the USB is on the top and WLAN antenna is on the bottom.
The main issue is the touch capabilities. The touch is registered in "some way" but only, where clicking in the upper half of the screen would scroll up in the list and clicking the lower half would scroll down. To enter a functionality you highlighted this way, You'd need to click right in the middle of the screen.
Also there is no way to get out of FLAP without resetting the device.

Maybe you are already on those issues and they will get adressed. Just wanted to share feedback.

Spooks4576, 2024-10-28:
> Feedback to compatibility: I am using an ESP32-24S032C CYD style board with a single USB-C ST7789 screen driver and GT911 touch driver. Basic functionality is fine as in the software will run and perform actions as expected, if you can manage to select your wanted option; see below. The screen could be flipped 180 degree. As is, the USB is on the top and WLAN antenna is on the bottom. The main issue is the touch capabilities. The touch is registered in "some way" but only, where clicking in the upper half of the screen would scroll up in the list and clicking the lower half would scroll down. To enter a functionality you highlighted this way, You'd need to click right in the middle of the screen. EDIT: The only area where touch is actually working on screen elements is in the main menu. Also there is no way to get out of FLAP without resetting the device.
> 
> Maybe you are already on those issues and they will get adressed. Just wanted to share feedback.

to exit flap on touch screens you click the center Game Over Button if you click outside of it it restarts the game 

ive set it up specifically so upper half of the screen is up lower half is down and center button is select / ok 

these are not bugs or issues but the way i have set things up

Moroyi, 2024-10-28:
My bad then, sorry. I did not see buttons but just the list itself filling the screen. I falsely assumed to click on the wanted option.

Spooks4576, 2024-10-28:
im going to close this issue as this is not a firmware bug but a thing to work on support in the future 

this will be announced when this specific CYD is ready

closed 2024-10-28 by Spooks4576