Spooks4576/Ghost_ESP #30: Issues with CYD 2USB (ESP32-2432S028R) firmware.

Colonel-Dixie, 2024-10-28:
Hello,       

Today I tried the firmware on CYD 2USB SP32-2432S028R, and faced issues. 

I downloaded the firmware https://github.com/Spooks4576/Ghost_ESP/releases/download/VA1.3/Screen_CYD2USB.zip 
I flashed it with https://esp.huhn.me/
Bootloader at 0x1000, partition at 0x8000, and the firmware at 0x10000, and after flashing, nothing at all on screen.      

Then I flashed the M5 launcher from the web flasher https://bmorcelli.github.io/M5Stick-Launcher/flash0.html
Then I installed the Ghost_ESP_IDF.bin within the launcher, and I got an image, but it's flipped, cropped, and the touch doesn't work. 
See attached video please:

https://github.com/user-attachments/assets/4743c152-e7b5-43ae-b0d9-0bdc5ce3b1d9

I also have a CYD variant 3.5" Capacitive [ESP32-3248S035C](https://www.aliexpress.us/item/3256804446638703.html) I can test if you want testers :)       
          
          
Thank you :)

Spooks4576, 2024-10-28:
what CYD do you have in specifically 

There are 3 variants im aware of 

- Micro USB (Supported)
- USB C (Supported)
- Micro USB and USB C (Not Supported)

Colonel-Dixie, 2024-10-28:
Oh, I just noticed that the one I got is not supported, I have one with 2USB, I thought it was supported since the file name is "Screen_CYD2USB.zip" 

[image]

I will close this issue as it's not a software issue, maybe a file name issue.   
Maybe changing the file name from "Screen_CYD2USB.zip" to "Screen_CYD_USB_C.zip might clear the confusion.     
 
Thanks again, and I hope see this project be ported to other devices too :)

closed 2024-10-28 by Colonel-Dixie

Spooks4576, 2024-10-28:
yep thank you so much i plan to add support for this CYD variant in November so stay on the look out