Spooks4576/Ghost_ESP #33: FAQ question - USB Serial port presentation

makergamer, 2024-10-29:
I did not see the answer to this question in the docs or FAQ:

Q: When connecting a Ghost_ESP board to a phone or PC without being in RST for a firware flash, does the USB present itself as a traditional Serial Port with a command prompt style interaction?

I know that [Marauder does](https://github.com/justcallmekoko/ESP32Marauder/wiki/cli). I did not want to assume that Ghost_ESP does also.

Personally, I would think many folks would be interested in a super tiny ESP32 having an microSD slot & USB-C **_MALE_** plug for warwalking with your phone, cableless mess with laptops & tablets, etc. I have yet to find one. I'd make one but lack the eagle cad and electrical skills for a custom build from JCLPCB or PCBWAY (with BOM assembly)

Spooks4576, 2024-10-29:
well that depends if you have a USB to UART Controller or a JTAG Controller on the board if not then you cant do it due to hardware limitations.

software wise its fully supported so long as this hardware is present.
stuff that wont work is the flipper dev board for example since flipper did not add this hardware to there dev board

Spooks4576, 2024-10-29:
note you can use USB-UART in the GPIO app to use serial with the ESP and PC

zen313-99, 2024-10-29:
> I did not see the answer to this question in the docs or FAQ:
> 
> Q: When connecting a Ghost_ESP board to a phone or PC without being in RST for a firware flash, does the USB present itself as a traditional Serial Port with a command prompt style interaction?

it works on PC, so it will work on the phone if proper usb-serial driver is installed.

Spooks4576, 2024-10-29:
im closing this as this is stale feel free to reopen this if you need further help

closed 2024-10-29 by Spooks4576

makergamer, 2024-10-29:
tyvm for the fast reply.