Spooks4576/Ghost_ESP #55: CYD dual/2 usb

CrackedGh0st, 2024-11-03:
Was confused which to use for 2.8 CYD with micro usb and type C. 
2USB- didnt work or the screen just didnt come on
Dualusb- does work but needs calibration as the touchs seem to have no rhyme or reason. Navagation is impossible. 
Completely understand its a work in progress and Im thankful for the work, just bringing it to attention. I can also test on a single micro usb CYD if need be but its apart of another project at the moment

Spooks4576, 2024-11-03:
can you be more specific of the touch has no rhyme or reason Navigation is indeed possible and i have a feeling your just not using it correctly or have something else going on

CrackedGh0st, 2024-11-03:
okay so ive now tested it on the micro usb only.
 the issue is the same for both CYD's
issue 1 - scrolling is not resposive and the only way to navigate is to tap on the lower or upper portion of the screen. 
issue 2 - selecting anything.. example. once in the wifi menu. You cant click on a tab and it open. instead you have to click to the middle and right of the screen.

ive seen you write off a few other people on here using the CYD screen as well. I wonder what has you so salty about it that you "have a feeling im not using it correctly" when you cant properly label your packages? 

are you testing any of these or just porting them over the best they'll go. You need to understand that there are alot of people and alot of variablies and getting pissy when someone opens a issues its going to make them go away. so if people asking real questions and seeking help is upsetting you. Close the project and keep it to yourself. Your high horse can still get you kicked it the balls

closed 2024-11-03 by CrackedGh0st