Spooks4576/Ghost_ESP #69: 2usc TC screen error

Pro123123kk, 2024-11-05:
The cYD version and the remaining versions have screen errors
[image]
[image]
[image]
[image]
[image]

Spooks4576, 2024-11-05:
just fixed the issue and updated bins on github and the flasher site for that varient please reflash 

please let me know if its fixed so i can close this issue

Pro123123kk, 2024-11-05:
It's been fixed, but running wifi and attack is not very good

Pro123123kk, 2024-11-05:
It's not completely finished yet, it's just beautiful and functional

Spooks4576, 2024-11-05:
> It's been fixed, but running wifi and attack is not very good

i recommend you flash again as i pushed a emergency update that fixes bluetooth and terminal view logs 

i will close this as the original issue is fixed please feel free to open a new issue if you encounter anything else

closed 2024-11-05 by Spooks4576

Pro123123kk, 2024-11-06:
Okay thank you for your help have a wonderful day