Spooks4576/Ghost_ESP #71: Does not support CYD ESP32-2432S024

Asmodeux2, 2024-11-06:
[image]
Stably works only on ESP32-2432S028 which is 2.8 inches.
On ESP32-2432S024 which is 2.4 inches - does not work

Spooks4576, 2024-11-06:
this is the USB C Version correct? it should work if you flash that variant

also if it doesn't work its gonna be hard for me to verify that any fixes i do would work since i don't have that CYD Model and don't plan on supporting it right now 

the only CYD Variants i support is 

Micro USB 
Dual USB 
And USB C 2.8 inch 

Maybe in the future ill begin supporting more CYDs but there's too many rn to count

Asmodeux2, 2024-11-06:
Buddy, I understand the confusion, but the ESP32-2432S024C module has a CST820 sensor. It simply won't work because this library is not declared. Please confirm that this firmware is only for the ESP32-2432S028.

Asmodeux2, 2024-11-06:
Не подтвердите а укажите в тексте чтобы другие не путались

closed 2024-11-06 by Asmodeux2

Spooks4576, 2024-11-06:
> Buddy, I understand the confusion, but the ESP32-2432S024C module has a CST820 sensor. It simply won't work because this library is not declared. Please confirm that this firmware is only for the ESP32-2432S028.

i plan to be more transparent in the future about what CYDs specifically work with this firmware thank you