Spooks4576/Ghost_ESP #78: Co.patibility witb ESP CYD

avipars, 2024-11-11:
Are there any CYD (cheap yellow displays) that are known to work well with ghostesp ?

avipars, 2024-11-11:
From other issues, it seems like usb c only CYDs work

Benefit of CYD is it had a display, microsd card reader, antenna built in. 

I will probably buy one and report back on progress and hopefully submit a pull request if I can add anything useful

Spooks4576, 2024-11-11:
> From other issues, it seems like usb c only CYDs work
> 
> Benefit of CYD is it had a display, microsd card reader, antenna built in.
> 
> I will probably buy one and report back on progress and hopefully submit a pull request if I can add anything useful

we only support 3 variants 

the micro USB 
The Dual USB With Micro and USB C 
and the Standard USB C

any other variants feel free to port yourself and make a pr thank you

closed 2024-11-11 by Spooks4576