Spooks4576/Ghost_ESP #85: Not work touch display on ESP32-2432S024C

Asmodeux2, 2024-11-15:
**You are a wizard**, you made the firmware for ESP32-2432S024(R), thank you very much. But can you make a firmware for **ESP32-2432S024(C)** it uses **CST820** library for touch screen. Thank you very much in advance!

Asmodeux2, 2024-11-15:
[image]

Asmodeux2, 2024-11-15:
[Library CST820 for touch display ESP32-2432S024C](https://github.com/codekulturbonn/espocto/blob/refs%2Fheads%2Fmain/src%2FCST820.h)

Spooks4576, 2024-11-15:
> [Library CST820.h for touch display ESP32-2432S024C](https://github.com/codekulturbonn/espocto/blob/refs%2Fheads%2Fmain/src%2FCST820.h)
> 
> [Library CST820.cpp for touch display ESP32-2432S024C](https://github.com/codekulturbonn/espocto/blob/main/src%2FCST820.cpp)

i can def add support for this however my only issue is i wont be able to confirm if it works or not without you contacting me on discord @ spooky4576 also this is because i am against releasing stuff without being able to verify it working first 

so if you really want this supported please contact me on discord thanks

closed 2024-11-15 by Asmodeux2

Asmodeux2, 2024-11-15:
You use Telegram?

Spooks4576, 2024-11-15:
> You use Telegram?

yes one sec