Spooks4576/Ghost_ESP #90: WebUI Evil Portal Start issue

Chaoscake, 2024-11-20:
Hello there...
When setting up Evil Portal in Web UI, the Start Evil Portal Button has no function.

Tested with Cardputer.

Spooks4576, 2024-11-20:
> Hello there... When setting up Evil Portal in Web UI, the Start Evil Portal Button has no function.
> 
> Tested with Cardputer.

that doesn't tell me anything does the evil portal access point get created upon pressing create?

by default it will not do anything frontend wise but on the backend it should work

Spooks4576, 2024-11-20:
this issue has been identified on my end and a fix has been pushed please wait 10 minutes for me to build the release again and then try to reflash using the flasher site or anything else thank you for bringing it to my attention