Spooks4576/Ghost_ESP #91: Flipper Hub Ultimate board Touchscreen issue

Chaoscake, 2024-11-20:
Hey again.
Updated the Flipper Hub Ultimate board with the spookytools flasher.
Touchscreen has no function. 
Flashed marauder on it to confirm functionality of the screen. Hardware is ok, Touchscreen works.
Flashed GhostESP 1.4.2 and 1.4.3 on it. 
Touchscreen with bot fw without function.

Spooks4576, 2024-11-20:
> Hey again. Updated the Flipper Hub Ultimate board with the spookytools flasher. Touchscreen has no function. Flashed marauder on it to confirm functionality of the screen. Hardware is ok, Touchscreen works. Flashed GhostESP 1.4.2 and 1.4.3 on it. Touchscreen with bot fw without function.

it should work as i verified it worked on 1.4.2 can you give me more details

Chaoscake, 2024-11-20:
I flashed as suggested with your flasher. FlipperHub, blue port, flashed successfully. Touchscreen doesn't work. Flashed Marauder with FZEE on it, Touchscreen works. Tried launching WebUi. I remembered it showing V 1.4.1 before, but now I cant even get the WebUi to show up.
I don't know what I'm doing wrong

Spooks4576, 2024-11-20:
> I flashed as suggested with your flasher. FlipperHub, blue port, flashed successfully. Touchscreen doesn't work. Flashed Marauder with FZEE on it, Touchscreen works. Tried launching WebUi. I [...]

id recommend you showing me some logs of you attempting to use it through the serial terminal as that will give me more insight of what is happening

Chaoscake, 2024-11-20:
[serial_logs.txt](https://github.com/user-attachments/files/17836796/serial_logs.txt)

Added logs... hope I did it right

Chaoscake, 2024-11-20:
SD card rendered fake touch input due to bord layout. SD card removed and Touchscreen worked. 
Unser error.
Thanks for the help though

closed 2024-11-20 by Chaoscake