Spooks4576/Ghost_ESP #95: t-watch-s3 screen not turning on

Howdyowen, 2024-11-25:
ti is just a black screen

Jupiter0016, 2025-03-07:
### Issue: SD Card Initialization Fails on ESP32-WROOM-32U

Hello, I need help with getting my SD card module to work with my ESP32-WROOM-32U. I’m using SPI mode, but the card **fails to initialize**. The serial monitor shows:

### **My Setup:**

- **Board:** ESP32-WROOM-32U  
- **SD Card Module:** [
https://a.aliexpress.com/_oovkOs2]  
- **SD Card Type & Format:** 4GB SanDisk, formatted as FAT32  
- **Connections (SPI Mode):**  
  - **VCC → 5V** 
  - **GND → GND**  
  - **MOSI → GPIO 23**  
  - **MISO → GPIO 19**  
  - **SCK → GPIO 18**  
  - **CS → GPIO 5**

Spooks4576, 2025-03-17:
this issue is stale please reopen a new one if you need further assistance

closed 2025-03-17 by Spooks4576