Spooks4576/Ghost_ESP #126: CYD failures, bug report section?

SoLongSidekick, 2024-12-21:
Is this where bug reports should be posted? Hope so.

So I've tried 2 different CYD boards, the 3.5" USB C version and the 2.8" dual USB version and neither worked. The 2.8" dual port version does seem to program successfully, but after booting it only shows 2 apps and a handful of other features while the 3.5" one fails to boot at all. 

Any sort of debugs I can run to provide the project with more information? Any known boards that provide even most of the functionality mentioned?

jaylikesbunda, 2024-12-21:
try getting any logs from the serial console at [spooky's website](https://spookytools.com/)

also are you flashing using the webflasher and the correct options for each CYD?

SoLongSidekick, 2024-12-22:
> try getting any logs from the serial console at [spooky's website](https://spookytools.com/)
> 
> also are you flashing using the [webflasher](https://flasher.ghostesp.net)? and the correct options for each CYD?

Yes sir, both flashed via web flasher and correct boards. I'll try to pull some logs to post here.

Also, I tried clicking the Discord link in the creator's account info here but it didn't take me anywhere. I'm a maker who just kind of loves helping with projects. I doubt my meager programming skills would be of any use, but I do have the ability to CNC my own PCBS extremely fast, so if you guys get tired of having to order multiple copies of prototype PCBS and just want to run a few off I'm always willing to help.

jaylikesbunda, 2024-12-22:
> Yes sir, both flashed via web flasher and correct boards. I'll try to pull some logs to post here.
>
> Also, I tried clicking the Discord link in the creator's account info here but it didn't take me [...]

as far as i'm aware the 3.5" CYD isn't supported as of current but you may be able to build a custom build for it depending on if you know the screen res and if it uses existing display driver, these links might help you out:
https://github.com/Spooks4576/Ghost_ESP/wiki/Board%E2%80%90Specific%E2%80%90Guide#supported-resolutions

https://github.com/Spooks4576/Ghost_ESP/wiki/Custom-Build-Guide#custom-build-using-github-actions

Spooks4576, 2025-03-17:
3.5 Inch is not supported and wont be supported for some time 

due to this i will be closing this issue. please open a new one if you have any other issues

closed 2025-03-17 by Spooks4576