Spooks4576/Ghost_ESP #128: Support for SHA2017 Badge (ESP32)

ifwg44, 2024-12-28:
In #118 i have been told to open a support request here.
I have a SHA2017 Badge i would like to use for something useful like for example running Ghost_ESP on it.
Here are the hardware details: https://badge.team/docs/badges/sha2017/hardware/

The hardware is as open build as possible. I hope this helps to add some support for it.

Spooks4576, 2024-12-31:
just FYI i probably wont start looking into this until we get most of the firmware done. There is still a lot to do 

not only that but i have no proper way of testing this as i dont have the badge. so even if i do port it over to this device. how am i supposed to verify that its working...

im not sure if these badges are being sold anywhere but if they are sending a link for me to purchase one would be even better and i would be more likely to port it over

Thank you for your understanding..

Spooks4576, 2025-01-02:
i mean if there is someway we can get in contact thats not github issues i might be able to work on adding support for it regardless if i have the device or not 

lemme know how you would prefer to stay in touch

closed 2025-04-17 by Spooks4576