Spooks4576/Ghost_ESP #129: Feature request: WPS PBC capture functionality

ifwg44, 2024-12-28:
Could you add support for the silent attack that is nearly nowhere implemented to decrypt the password by a pushed WPS button?

Pushing a button that temporary disable the security of a network is always a bad idea. I try to demonstrate this to people but there is still no simple, power efficient solution for that.

The idea is out there for a longer time: https://github.com/FattusRattus/WPS-PBC-Truffle

Could you add this to Ghost_ESP? Running a ESP32 24/7 for endless time waiting for someone to press the WPS button is one of the most energy efficient ways for this. Also the name of this project fits perfectly. Its the silent ghost that is living unseen listening the whole time for someone to press the WPS button to reveal the WPA PSK passphrase.

Spooks4576, 2024-12-31:
i could implement this. however im not sure if the ESP-IDF Supports the pin method in newer versions of its framework 

when i was messing around with WPS before i found out that they removed or deprecated PIN Method support after a certain version requiring me to edit the ESP IDF Source Code directly to expose variables and functions that would normally not be accessible in ESP IDF 

that being said it is still possible to add the pin method back through other means however that would require me to to tweak the WPS Implementation that is currently in ESP IDF 

that also being said doing it myself and remaking the entire thing from scratch would be not feasible due to how complicated the handshakes can get (I've looked into the source for the current WPS Implementation it is far from simple)

maybe in the future i can add this via forking the ESP IDF branch and building off of that but until then ill leave this open as a later possibility

Spooks4576, 2025-01-02:
after relooking at it again yes it is possible however this feature would likely be something i would add later down the line 

i shall keep this issue open until this has been added

closed 2025-04-22 by Spooks4576