Spooks4576/Ghost_ESP #135: Idea: 802.15.4 support for ESP32-C6

battlehax, 2025-01-08:
It would be neat to be able to scan, sniff, spam, and or spoof IoT devices using Zigbee and Matter protocol.
I believe that it's right in line with the goals and other features of this project.
It appears that espressif has libraries available to use these protocols, but I'm unable to get their examples to run reliably yet.

detonation, 2025-02-01:
Would be neat, even if you have to run 2 esp's back to back, and have the c6 for zb/matter, and main esp for rest. Kinda like CYD and how they do the crazy e-paper tag setups (), where they have a s3 for main proc, c6 for zigbee/matter, cc1101 for rf, all in one device. https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Building-an-Access-Point for example, with different firmware for the c6

Spooks4576, 2025-04-17:
this is more hardware than software and currently out of scope for this project so im sorry but it wont happen

closed 2025-04-17 by Spooks4576

detonation, 2025-04-18:
Fair enough, I didn't mean to imply I was unhappy with your work - it's astounding. Thanks!