Spooks4576/Ghost_ESP #139: New "apcred" feature doesn't work on ESP32-S3-Wroom-1

byFynn, 2025-02-04:
I saw the new apcred feature in change log tried it and it didn't work on my ESP32-S3-Wroom-1 w/o MicroSD Card. I mean it should work I guess because when u type help to see all commands there is no line that says "MicroSD needed" like on other commands or sth like that...

The command output first says that like 2 things failed and than after that it says it changed it. But it didn't do anything so AP name and pw still same.

Spooks4576, 2025-02-07:
this feature is honestly a hit or miss cause i have a ESP32-S3-WROOM on a waveshare board and ive tested and verified it works on everything except the C3 series 

i will do my best to fix this and keep you updated if i find any solutions

jaylikesbunda, 2025-02-07:
> this feature is honestly a hit or miss cause i have a ESP32-S3-WROOM on a waveshare board and ive tested and verified it works on everything except the C3 series 
> 
> i will do my best to fix this and keep you updated if i find any solutions

we could force it to run off the custom creds even if it doesn't save to nvs i'm assuming it's something to do with nvs

closed 2025-03-06 by jaylikesbunda