Spooks4576/Ghost_ESP #142: please help me

munirhussain2872001, 2025-03-03:
This is my board info https://github.com/rzeldent/platformio-espressif32-sunton?tab=readme-ov-file#esp32-2432s022-nc Please write a program for this board because you are the best coder but I am not 🥺🥺🥺🥺🥺😥😥😥😥😥😥😥😥😥 plz plz plz    

CYD-2432S022C (Capacitive)

munirhussain2872001, 2025-03-03:
[image]           only its work 😥😥😥😥😫😫😫😫

munirhussain2872001, 2025-03-04:
Thank you sir you are a great person.😍😘🥰❤💗💖

Spooks4576, 2025-03-17:
this board should be supported under the 

[image]

build please flash this and reopen a new issue if your still having problems thank you

closed 2025-03-17 by Spooks4576

munirhussain2872001, 2025-03-19:
Sir, it doesn't work after flashing just the white screen.  please help me 😥😥😥😥

munirhussain2872001, 2025-03-19:
> this board should be supported under the
> 
> [image]
> 
> build please flash this and reopen a new issue if your still having problems thank you
                                                                               CYD-2432S022C (Capacitive)  my bord sir please help me