Spooks4576/Ghost_ESP #145: WebUI Credentials

CeresF3b, 2025-03-09:
I believe that GhostEsp should require login credentials to access the WebUI—not just the Wi-Fi password, but actual user credentials (username and password) specifically for the WebUI. The reason for this is that if the ESP is connected to a Wi-Fi network and someone scans that network, they could see that a device has an open port, which would be the WebUI. A smart attacker could then access it and see what’s happening. I think this additional security layer is absolutely **necessary**.

[image]

Spooks4576, 2025-03-09:
i will definitely add this into consideration for 1.4.8 and have it be configurable 
shouldn't be too hard either 
stay tuned for updates

CeresF3b, 2025-03-09:
> i will definitely add this into consideration for 1.4.8 and have it be configurable shouldn't be too hard either stay tuned for updates

Thanks dude :)

Spooks4576, 2025-04-17:
this has been added and is done

closed 2025-04-17 by Spooks4576