Spooks4576/Ghost_ESP #146: ESP32-2432S028R Dual USB Unsupported?

mrapxs, 2025-03-11:
Tried ESP32-2432S028R and there was some screen issues, couldn't see any back button or anything, no exits for UI's or applications, and none of the WiFi or BLE functions worked at all. Evil Portal showed the AP but disallowed connections. I know it's correct hardware because I flashed every other version of GhostESP unsuccessfully

jaylikesbunda, 2025-03-11:
> Tried ESP32-2432S028R and there was some screen issues, couldn't see any back button or anything, no exits for UI's or applications, and none of the WiFi or BLE functions worked at all. Evil Portal [...]

most of your 'issues' are the design, most screens you can use top left corner to go back. not sure what happened with evil portal but there should be back buttons next update thanks to spooky

mrapxs, 2025-03-15:
Hopefully there's back buttons in the terminal as well, because I can exit menus in the top left corner but I have to restart my CYD Dual USB to be able to exit terminals

Spooks4576, 2025-03-17:
evil portal needs a rework 

as for the screen, everything should be functional though ill make sure to add more visual elements for going back for different views

closed 2025-04-22 by Spooks4576