Spooks4576/Ghost_ESP #147: Add support for ESPBOY

coppaxamsukk, 2025-03-12:
Espboy is a diy esp8266 based project that's similar to cardputer but available from long ago. 

Can you port the awesome project for espboy?
https://www.espboy.com/

[image]

Spooks4576, 2025-03-17:
sadly because this is based on the esp8266 i will not be supporting this since this project is only compatible with ESP32 modules not the esp8266

Thanks for your understanding

closed 2025-03-17 by Spooks4576

coppaxamsukk, 2025-03-17:
> sadly because this is based on the esp8266 i will not be supporting this since this project is only compatible with ESP32 modules not the esp8266
> 
> Thanks for your understanding

Ohh, your project is awesome.
However, espboy will release esp32boy based on esp32, i will let you know.
Please then port ❤️

.
https://community.espboy.com/t/esp32boy-the-next-advanced-espboy/300/9